jmplonka/InventorLoader #1: Add as Addon to FreeCAD-Addons as experimental Alpha

luzpaz, 2017-05-18:
Please consider adding this in its alpha state to the freecad-addons so it can get more visibility and testers, when you are ready for feedback. I'm sure you can state your timeframe, 2years, and working at your own pace to impement features as you go, AKA you are in no rush. What do you think?

jmplonka, 2017-06-13:
Thank you for the feedback. 
I think with 2D-Sketches and "extrusion" (ak. "pad") feature, the addon will be ready for phase "alpha".
Stating a timeframe is a bit more tricky, because:
- [ ] boolean-,
- [ ] edge- (phase, fillet),,
- [ ] loft & sweep-,,
- [ ] mirror- and pattern-features
are "planned" to be available with the "beta" version.

luzpaz, 2017-06-26:
Fantastic!

jmplonka, 2017-09-06:
It should be now available :)
But I've no idea of how to set status info (alpha-state).
BTW a new Feature has been added: "Holes"

closed 2017-09-06 by jmplonka