jmplonka/InventorLoader #13: Maximum recursion depth exceeded in cmp in build()

markol, 2018-06-23:
Problem occurs with some specific files (sample [test.zip](https://github.com/jmplonka/InventorLoader/files/2130231/test.zip) enclosed). Setting recursion limit does not help, fast temporary fix is possible by adding counter variables to break recursion, like:

```
maxRecursion = 500
...
global counter
	if (self.type == 'ref' and counter < maxRecursion):
``` 

```
  File "~/.FreeCAD/Mod/InventorLoader/Acis.py", line 2857, in build
    self.shape = self.surface.build()
  File "~/.FreeCAD/Mod/InventorLoader/Acis.py", line 2857, in build
    self.shape = self.surface.build()
  File "~/.FreeCAD/Mod/InventorLoader/Acis.py", line 2857, in build
    self.shape = self.surface.build()
  File "~/.FreeCAD/Mod/InventorLoader/Acis.py", line 2857, in build
    self.shape = self.surface.build()
  File "~/.FreeCAD/Mod/InventorLoader/Acis.py", line 2857, in build
    self.shape = self.surface.build()<type 'exceptions.RuntimeError'>: maximum recursion depth exceeded in cmp```

jmplonka, 2018-06-26:
Hi!
Thank you for sharing the sample file.
The sat files contains references to subtypes. Currently there is a bug in building up the ref-table. So that sometimes there can be endless loops.
I'm working on it but can't find a correct implementation.
As a workaround I will simple try to detect and break these "self-references".
-Jens

markol, 2018-06-26:
I agree that more investigation is needed for that case. Any workaround I have tried, results in surface being omitted in imported solid.

jmplonka, 2018-06-29:
The omitted surfaces should now be fixed.

jmplonka, 2018-07-02:
Closed: issue fixed, whereas the subtype table for surfaces is nevertheless incomplete.

closed 2018-07-02 by jmplonka