jmplonka/InventorLoader #16: Can't open *.ipt file

andy-kliman, 2018-08-22:
When I try to open * .ipt, I get the following:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Andy\Desktop\PORTABLE APPS\PortableApps\FreeCAD Portable\App\FreeCAD\Mod\InventorLoader\importerIL.py", line 8, in <module>
    import os, sys, FreeCAD, importerSAT, Import_IPT
  File "C:\Users\Andy\Desktop\PORTABLE APPS\PortableApps\FreeCAD Portable\App\FreeCAD\Mod\InventorLoader\importerSAT.py", line 11, in <module>
    from Acis2Step     import export
<class 'SyntaxError'>: invalid syntax (Acis2Step.py, line 637)
```
FreeCad version 0.18 build 14426 (Git)

jmplonka, 2018-08-30:
Can you please send me the file to jens.m.plonka[at]gmail.com?

andy-kliman, 2018-09-01:
I downloaded freecad without Conda and everything was opened, but the model itself was not visually displayed

jmplonka, 2018-09-02:
The SAT part is still buggy - sorry!
Please use as a workaround the convert to STEP feature.

jmplonka, 2018-09-03:
BTW - which version (date) do you use?
I wasn't able to reproduce your traceback :(

jmplonka, 2018-09-03:
another workaround for nativ import - disable dimension constraints as they are buggy!
Open user.cfg of FreeCAD (on windows: %APPDATA%\FreeCAD\user.cfg and search for "Sketch.Constraint.Dimension" and set all values to "0".
Radius and Diameter you can set to "1" as they normally work.

andy-kliman, 2018-09-03:
That traceback I got in FreeCad version 0.18 build 14426 Conda_Py3QT5 Win10. But I opened with the latest github build without Conda.

When I opened my file at the same time in Inventor and Freecade, I compared the sketches and saw that some were slightly changed in the freeCAD, in addition there is no chamfer at all.
Also i dont understand how to install libs on mac to use InventorLoader

jmplonka, 2018-09-04:
Chamfer, like any other operation is not yet implemented - I'm sorry for that, but I'm working on it.

Unfortunately, I don't have tested InventorLoader on a Mac. So please ask in the FreeCAD forum for help on installing the required packages as they are not installed automatically.

jmplonka, 2018-09-14:
IL should now also run with python 3 from now on.

andy-kliman, 2018-09-20:
Ok, i'll try

jmplonka, 2018-10-11:
I just finished creating a separate table for iParts when importing IPT files in native model.

Please  check, if this is more likely what you've expected.

closed 2020-05-20 by jmplonka