jmplonka/InventorLoader #18: Logo for this workbemch

luzpaz, 2018-10-04:
Ref: https://www.freecadweb.org/wiki/Template:DevWorkbenches
Does it also have a page on the wiki? If so please add it to https://www.freecadweb.org/wiki/External_workbenches

jmplonka, 2018-10-10:
BTW: InventorLoader is a plugin - not a workbench. It comes with no additional UI.
I can't see any need for an icon, but there's already one in ./Resources/icons/InventorImporter.svg
Until now it has no wiki page right now that could be added to FC's  external workbenches page (because it is no workbench).

luzpaz, 2018-10-10:
Good point, though we need to represent it somehow. We'll add it to the wiki page with a notation about the discrepency

luzpaz, 2018-10-12:
@jmplonka We still need a way to represent your plugin in the Addon Manager and wiki. I see there is a https://github.com/jmplonka/InventorLoader/blob/master/Resources/icons/InventorImporter.svg 
I could also post on the FC UI/UX subforum to see if anyone wants to create a logo for this...or do you prefer to roll your own? what do you think ?

jmplonka, 2018-10-12:
There is one Point, that might be worth to be available in FreeCAD that is a common feature in Autodesk Inventor: "iParts"
With iParts you are able to define a set of variations of a model. The values are held in a table. An iPart defines a set of parameters that are differently set by each variant model.
With this Feature InventorLoader would be more like a WorkBench than.
ps: Any support, that will be offered to me, is highly appreciated.

jmplonka, 2019-03-06:
Well, finally InventorLoader got it's Workbench with "Inventor" menu and toolbars for:
* [create new objects](../wiki/Toolbar-Create)
* [modify existing objects](../wiki/Toolbar-Modify)
* [modify surfaces](../wiki/Toolbar-Surface)
* [apply patterns](../wiki/Toolbar-Patterns)
* [handle Sheet-Metal](../wiki/Toolbar-SheetMetal)
* [handle Plastic](../wiki/Toolbar-Plastic)

I've also added a page for each already available features. Some of the features will obviously simply run the functionality other workbenches offers (e.g. extrude, revolve, ...)

You've asked for an [icon](../blob/master/Resources/icons/Inventor-WB.svg) - does it meets the requirements?

closed 2019-03-06 by jmplonka