jmplonka/InventorLoader #19: Some questions about how plugin works

nicedayvl, 2018-10-14:
Hi, could you tell me please how plugin works? Where are the dots, lines and other primitives imported in from what stream are they read and where are they resides in python code. Basically i would like to help with development of the greate plugin but need some info on plugin design. Thanks in advance.

jmplonka, 2018-10-14:
Thank you for the compliment.
I worried, that one day someone will come to get some information about how the plugin is working ,>
Please give me some time, so that I can write some documentation in the Wiki. Till than I will try to give you a short Sketch about the structure... please be patient, I will write you soon.

jmplonka, 2018-10-17:
I just updated the FileFormat.odt in [DOCU](https://github.com/jmplonka/InventorLoader/blob/master/Docu/FileFormat.odt)

jmplonka, 2018-10-17:
And now the first draft of the [workflow](https://github.com/jmplonka/InventorLoader/blob/master/Docu/Workflow.htm)...

nicedayvl, 2018-10-17:
Greate job, much appreciated! I will look at this all and write back soon.

nicedayvl, 2018-10-18:
Could you explain please more in details how buffer format of Point2D, Circle2D, Line2D, etc primitives
And what exactly procedure like does:

	def ReadSketch2DEntityHeader(self, node, typeName):
		node.typeName = typeName
		i = self.ReadHeadersss2S16s(node)
		i = node.ReadCrossRef(i, 'refSketch')
		return I

what does ReadHeadersss2S16s exactly?

jmplonka, 2018-10-18:
Well, `ReadHeadersss2S16s` is a structure at the beginning of many section nodes.
Most sections have a `UInt32` + `UInt16` 'main' header (dont't know anything about the meaning).
Inventor before 2011 stores additionaly the size from time to time.
For the 'main' header this is 0x00000006 (4 Bytes for an UInt32 + 2 Byte for the UInt16).

Content-Header 2S16S(`ReadHeadersss2S16s`)
* The content header 3s
* second set of flags (`UInt32`), in the early versions this were 2 `UInt16` -> 2S16
* optional Size

Content-Header 3S (`ReadCntHdr3S`)
* The content-Header
* 3 x optional Size (`3S`)
* optional `UInt3` for Inventor version > 2018

Content-Header (`ReadContentHeader`):
* Main-Header (`UInt32` + `UInt16` + optional-Size)
* Reference number to another section node (`UInt32`)
* Flags `flags` for the content (`Uint32`)
* optional Size
* Refefrence number to the (parent) section node (`UInt32`)
* The indexNumber `index` (not index) of the node (`UInt32`)

BTW: I use Visusl Studio for the development.

closed 2018-10-18 by jmplonka