jmplonka/InventorLoader #20: Guide how to setup MS Visual Studio for debugging add-on would be very helpfull

nicedayvl, 2018-10-19:
hi Jens, could you share please your knowledge how to configure MS Visual Studio, say 2015 version to run and debug your addon inside FC

is that possible?
thanks in advance

nicedayvl, 2018-10-19:
step by step guide would be very helpfull btw

jmplonka, 2018-10-22:
I'm working with to development version to save time:
* one to run on commandline to validate correct syntax
* one to run with FreeCAD to validate correct working 

After synchronizing the FreeCAD version with the command line version I do normally a push to git if everything is working as expected.
1. in VS select new project -> Intalled/Templates/Python: "Python Application"
  a) define a name (e.g.: "`InventorLoader_cmd`")
  b) set the location (e.g.: "`D:\Projects\Python`") - the subfolder will be created for your project
  c) uncheck "Create directory for solution"
2. Remove the default python file because be we don't need it.
3. Drag'n drop all required files into the view "Solution Explorer", alternatively you can add already existing items by pressing `Shft+Alt+A`
4. Define _main.py as default startup script (context menu: "Set as Startup file")
5. Add the missing wrapper files after removing the .txt extension:
  * [_main.py](https://github.com/jmplonka/InventorLoader/files/2501593/_main.py.txt)
  * [App.py](https://github.com/jmplonka/InventorLoader/files/2501595/App.py.txt)
  * [Draft.py](https://github.com/jmplonka/InventorLoader/files/2501594/Draft.py.txt)
  * [FreeCAD.py](https://github.com/jmplonka/InventorLoader/files/2501610/FreeCAD.py.txt)
  * [FreeCADGui.py](https://github.com/jmplonka/InventorLoader/files/2501596/FreeCADGui.py.txt)
  * [PartDesign.py](https://github.com/jmplonka/InventorLoader/files/2501597/PartDesign.py.txt)
  * [Part.py.gz](https://github.com/jmplonka/InventorLoader/files/2501617/Part.py.gz) -> must be  extracted!
  * [Surface.py](https://github.com/jmplonka/InventorLoader/files/2501592/Surface.py.txt)
  * [Sketcher.py](https://github.com/jmplonka/InventorLoader/files/2501600/Sketcher.py.txt)
  * [Spreadsheet.py](https://github.com/jmplonka/InventorLoader/files/2501601/Spreadsheet.py.txt)
6. By default VS sets up the Project with the global defined python version. Here you can specify your own / different installation by adding a new python environment (e.g.: Python 3.7) via it's context menu: `Add/remove python environments`

nicedayvl, 2018-10-23:
hi Jens, I did al step by step and getting exception in importerUtil.py

from PySide.QtCore import *
from PySide.QtGui  import *

any ideas?

nicedayvl, 2018-10-23:
exception says DLL load fialed

jmplonka, 2018-10-23:
It seems, you have already an QT installation for python. Unfortunately it is not compatible with your python version (e.g. 32Bit vs 64bit) linked in the project.
Maybe you can try to install a new Python version (e.g. 3.7) somewhere and reference it in your InventorLoader project (by adding this python environment).
But don't forget to install the required libraries (open the `cmd.exe` and `set PATH=somewhere\bin` and navigate to the libs folder and run `python installLibs.py`!

nicedayvl, 2018-10-24:
I have no QT installation, I have no QT at all besides QT dlls in bin folder of FC

nicedayvl, 2018-10-24:
seems I solved problem with QT, now other exception No module named BOPTools.GeneralFuseResult in Acis.py

ImportError was unhandled by user code
Message: No module named BOPTools.GeneralFuseResult

nicedayvl, 2018-10-24:
btw I set directory C:\Program Files\FreeCAD 0.17\bin\ very first in the system wide PATH

jmplonka, 2018-10-24:
Can you please try to extract into project folder:
[BOPTools.zip](https://github.com/jmplonka/InventorLoader/files/2511422/BOPTools.zip)

nicedayvl, 2018-10-24:
thanks! but I copied files into project dir and added to workspace and still get same exception like python don't see them

nicedayvl, 2018-10-24:
Running D:\Projects\Python\InventorLoader_cmd\_main.py
Traceback (most recent call last):
  File "D:\Projects\Python\InventorLoader_cmd\_main.py", line 4, in <module>
    import importerIL, importerSAT, importerUtils
  File ".\importerIL.py", line 8, in <module>
    import os, sys, FreeCAD, importerSAT, Import_IPT
  File ".\importerSAT.py", line 8, in <module>
    import tokenize, sys, FreeCAD, Part, re, Acis, traceback, datetime, ImportGui
  File ".\Acis.py", line 12, in <module>
    from BOPTools.GeneralFuseResult import GeneralFuseResult
ImportError: No module named BOPTools.GeneralFuseResult

nicedayvl, 2018-10-24:
would you like to try teamviewer session please
do you have skype btw?

jmplonka, 2018-10-24:
It's a little bit tricky to analyse the issue.
your project folder should now look like:
`
...
InventorLoader_cmd\Acis.py
BOPTools\__init__.py
BOPTools\GeneralFuseResult.py
BOPTools\Utils.py
...
`
Your project should in VS also look like:
[image]

nicedayvl, 2018-10-24:
now missing ImportGui

nicedayvl, 2018-10-24:
and I haven't folder PySide and its contents btw

jmplonka, 2018-10-24:
[PySide.zip](https://github.com/jmplonka/InventorLoader/files/2511889/PySide.zip)
Sorry for all these inconveniences...

nicedayvl, 2018-10-24:
ImportGui please :)

nicedayvl, 2018-10-25:
I wrote you mail to your gmail address from metadata.txt

nicedayvl, 2018-10-25:
[image]

jmplonka, 2018-10-25:
The reason is, that in your `%APPDATA%\FreeCAD\user.cfg` a value is missing:
`					    <FCBool Name="checkLogging" Value="0"/>`
Please replace line 62 in FreeCAD.py with:
`e = xml.etree.ElementTree.SubElement(self.prm, 'FCBool')`

nicedayvl, 2018-10-25:
thanks Jens! now debugging seems working as needed

closed 2018-10-25 by jmplonka