jmplonka/InventorLoader #25: What's the status of this Addon ?

luzpaz, 2019-06-04:
Just wondering what milestone this is at on the roadmap...?

jmplonka, 2019-06-24:
.. what road map?
To be honest: my first ideaa was, to implement all features (more than 40). But unfortunately I came to a point where I must admit, that some of the fatures will only be very hard be implemented in FreeCAD as other AddOns would be requried (e.g. FEM or Sheet-Metal).
Therefore I decides to use the BREP Reperesentation for these features. The test looks quite well right now.
Currently I'm checking the side-effects, before I release my first 1.0 Version (Aug. 2019)
The next steps would be support for Drawings (2.0), Assemblies (3.0) and Presentations (4.0).
Until now I didn't got much feedback how usefull this addon/workbench is for the community.

luzpaz, 2019-06-24:
It needs some documentation and proof of concepts so that the people can see its utility

jmplonka, 2019-06-24:
do you have an example for documentation and/or proof of concept?
For the demonstration of the addon, I added some [demos](https://github.com/jmplonka/InventorLoader/tree/master/Demo-Status)

sslupsky, 2020-01-25:
Does it work with Fusion 360 (.f3d) files?

jmplonka, 2020-01-26:
No. Currently there are no plans to support F3D.

jmplonka, 2021-03-09:
Correction: it now supports AUTODESK Fusion360 files!

jmplonka, 2021-03-09:
@luzpaz: I think a first milestone is reached!
* Support of IPT, DXF, F3D and ACIS files.
* Most of the featues are implemented that are supported by FC (except metal sheets)
* Next step is to finalize the documentation.

luzpaz, 2021-03-09:
> Most of the featues are implemented that are supported by FC (except metal sheets)

/cc @shaise @jaisekjames  
Maybe we could extend sheetmetal wb to accommodate ?

luzpaz, 2021-03-09:
> I'm getting a little bit nervous about SheetMetal, because of missing experience! ;o>
> Neither in Invetor nor in FC!

https://forum.freecadweb.org/viewtopic.php?f=24&t=49546

jmplonka, 2023-05-02:
next milestones:
* make Preference-Page working for IPT-Import
* implement Sheet-Metal features.