jmplonka/InventorLoader #28: how to disable dimension constraints

sibbi77, 2020-08-23:
The documentation says:

"Please disable Dimension constraints in user.cfg:
```
<FCBool Name="Sketch.Constraint.Dimension.Angle2Line" Value="0"/>
<FCBool Name="Sketch.Constraint.Dimension.Angle3Point" Value="0"/>
```
"

but where should I add these lines?
The user.cfg is an xml file with, among others:
group: `<FCParamGroup Name="Sketcher">`
and within this group: `<FCParamGroup Name="General">`
Nothing is spelled like "Sketch.Constraint.[...]"

jmplonka, 2021-01-04:
Thank you reminding me, that the documentation is not synchronous with the implementation.

Initially it was planned to allow the user to enable or disable constraints in the GUI. Until now I wasn't able to do the implementation, so the preferences got really working.
Up to now, whatever is written in the user.cfg file is not interpreted.
If someone can help me out with making the properties/ImportExport/IPT stored in the user.cfg file this is no longer necessary!
Best regards
Jens

luzpaz, 2023-04-26:
@AjinkyaDahale would you have a moment to weigh-in ?

AjinkyaDahale, 2023-04-26:
@luzpaz what am I looking at here?

sliptonic, 2023-04-26:
This is interesting but the issue needs improvement.  The origin seems to be several years old.  There's no link to a forum topic or even the documentation in question.

luzpaz, 2023-04-27:
It's in the README:  https://github.com/jmplonka/InventorLoader#constraints-in-native-ipt-import

jmplonka, 2023-05-02:
@luzpaz : where can I find an example of how to store user setting from the Preference-Dialog? I want to implement the missing part for Import-Export settings for IPT file types.

jmplonka, 2023-05-02:
Is it sufficient (ref. SheetMetal preferences) to add simply:
```XML
<property name="prefEntry" stdset="0"><cstring>Sketch.Constraint.Geometric.AlignHorizontal</cstring></property>
<property name="prefPath" stdset="0"><cstring>Mod/InventorLoader</cstring></property>
```

jmplonka, 2023-05-03:
I think, I found the trick by replacing `QCheckBox` by `Gui::PrefCheckBox`:
```XML
                      <widget class="Gui::PrefCheckBox" name="cbxDimOffsetSpline">
                       <property name="text"><string>offset spline</string></property>
                       <property name="icon"><iconset><normaloff>../icons/Constraint_Length_Driven.svg</normaloff>../icons/Constraint_Length_Driven.svg</iconset></property>
                       <property name="checked"><bool>false</bool></property>
                       <property name="toolTip"><string>Import "offset spline" dimensions in 2D and 3D sketches.</string></property>
                       <property name="prefEntry" stdset="0"><cstring>Sketch.Constraint.Dimension.OffsetSpline</cstring></property>
                       <property name="prefPath" stdset="0"><cstring>Mod/InventorLoader</cstring></property>
                      </widget>
```