jmplonka/InventorLoader #29: Failing on Import/Open IPT file.

KeithSloan, 2021-02-20:
Opening an IPT file, get a mass of assertion errors before an error that aborts.
FreeCAD Report View has
`
16:37:12  Reading: /Users/keithsloan/Downloads/Gob-angle-corner-b-V2.ipt
16:37:13      created with Autodesk Inventor 2008 (Build 0)
16:37:13  	'Inventor Document Summary Information': (LC = 813)
16:37:13  	'Design Tracking Properties': (LC = 813)
16:37:13  	'_Private Model Information': (LC = 813)
16:37:13  	'Design Tracking Control': (LC = 813)
16:37:13  	'Inventor User Defined Properties': (LC = 813)
16:37:13  	'Inventor Summary Information': (LC = 813)
16:37:29   1: 'cyuvachsvraj3mcceewzr2ivjh' ('PmGraphicsSegment')
16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerGraphics.py", line 269, in Read_14533D82
    i = node.ReadUInt8A(i, 3, 'a3')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 234, in ReadUInt8A
    x, i = getUInt8A(self.data, offset, n)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 441, in getUInt8A
    assert end <= len(data), "Trying to read UInt8 array beyond data end (%d, %X > %X)" %(size, end, len(data))
AssertionError: Trying to read UInt8 array beyond data end (3, 72 > 71)

16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerGraphics.py", line 269, in Read_14533D82
    i = node.ReadUInt8A(i, 3, 'a3')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 234, in ReadUInt8A
    x, i = getUInt8A(self.data, offset, n)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 441, in getUInt8A
    assert end <= len(data), "Trying to read UInt8 array beyond data end (%d, %X > %X)" %(size, end, len(data))
AssertionError: Trying to read UInt8 array beyond data end (3, 6E > 6D)

16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerGraphics.py", line 269, in Read_14533D82
    i = node.ReadUInt8A(i, 3, 'a3')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 234, in ReadUInt8A
    x, i = getUInt8A(self.data, offset, n)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 441, in getUInt8A
    assert end <= len(data), "Trying to read UInt8 array beyond data end (%d, %X > %X)" %(size, end, len(data))
AssertionError: Trying to read UInt8 array beyond data end (3, 6E > 6D)

16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerGraphics.py", line 275, in Read_2C7020F6
    i = node.ReadUInt8A(i, 3, 'a3')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 234, in ReadUInt8A
    x, i = getUInt8A(self.data, offset, n)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 441, in getUInt8A
    assert end <= len(data), "Trying to read UInt8 array beyond data end (%d, %X > %X)" %(size, end, len(data))
AssertionError: Trying to read UInt8 array beyond data end (3, 3E > 3D)

16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerGraphics.py", line 92, in Read_60FD1845
    i = node.ReadUInt32(i, 'u32_1')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 264, in ReadUInt32
    x, i = getUInt32(self.data, offset)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 521, in getUInt32
    val, = UINT32(data, offset)
struct.error: unpack_from requires a buffer of at least 309 bytes for unpacking 4 bytes at offset 305 (actual buffer size is 305)

16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
[... 1727 lines cut ...]
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerBrowser.py", line 664, in Read_F7676AB2
    i = node.ReadUInt32(i, 'u32_2')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 264, in ReadUInt32
    x, i = getUInt32(self.data, offset)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 521, in getUInt32
    val, = UINT32(data, offset)
struct.error: unpack_from requires a buffer of at least 114 bytes for unpacking 4 bytes at offset 110 (actual buffer size is 110)

16:37:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegment.py", line 578, in HandleBlock
    i = readType(node)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerBrowser.py", line 664, in Read_F7676AB2
    i = node.ReadUInt32(i, 'u32_2')
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerSegNode.py", line 264, in ReadUInt32
    x, i = getUInt32(self.data, offset)
  File "/Users/keithsloan/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 521, in getUInt32
    val, = UINT32(data, offset)
struct.error: unpack_from requires a buffer of at least 126 bytes for unpacking 4 bytes at offset 122 (actual buffer size is 122)

16:37:29   6: 'xvcixoacuk2efdw4f4nh1ryufo' ('NBNotebookSegment')
16:37:29  Dumped data to folder: '/Users/keithsloan/Downloads/Gob-angle-corner-b-V2_ipt'
16:37:29  DONE!
[Gob-angle-corner-b-V2.ipt.zip](https://github.com/jmplonka/InventorLoader/files/6015666/Gob-angle-corner-b-V2.ipt.zip)

`

jmplonka, 2021-03-04:
Sorry but min Version is 2010!
It is not very clearly mention on the readme page.

closed 2021-03-04 by jmplonka

KeithSloan, 2021-03-05:
Given you the first two lines output to the console are
16:37:12 Reading: /Users/keithsloan/Downloads/Gob-angle-corner-b-V2.ipt
16:37:13 created with Autodesk Inventor 2008 (Build 0)

Could you not add a check for the version and put out a message.
This file was created with version xxxx, only versions 2010-2019 are supported
Then exit cleanly from the import.

ricardodeazambuja, 2025-02-05:
Would you have more info on the differences between Autodesk Inventor 2008 and 2010? Thanks!

jmplonka, 2025-02-05:
Hi,
till version 2009 Inventor "hides“ this info a little bit and the data blocks had can have less/extra/other data - hard to decode.