jmplonka/InventorLoader #33: Not filling in the body of IPT file

gjpc, 2021-03-15:
Thank you @jmplonka for your work on InventorLoader. I hope you have time and inclination to work on this issue. I have installed the loader on both .18 and .19

The following object 
[image]

Converts to

SAT:
[image]

nativ:
[image]

Convert to STEP simply fails.

Here is the ipt file zipped

[Gear 104T.zip](https://github.com/jmplonka/InventorLoader/files/6143878/Gear.104T.zip)

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

and FreeCAD .19

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.8
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

jmplonka, 2021-03-15:
Hello, sorry to hear that you have problems, while using the converter.
Thank you very mucht for sharing the example, so that I can have a look into it.
Till soon
Jens

jmplonka, 2021-03-16:
I have a general problem with generating the face for a surface and the wires. Sometimes it's not possible for me to calculate the remaining face from the set of wires and the given surface - I must regrett, but there will no solution in the nearer future for that.
The STEP is working (Not as pretty as in Inventor but not so far away):
[image]

jmplonka, 2021-03-16:
Should be fixed now - please verify:
[image]

gjpc, 2021-03-16:
**Verified!** Thanks @jmplonka ! Looks like the gears and supports are importing nicely under STEP!

I really don't want to be a buzzkill, but the back plate of the [Antikyra ](https://grabcad.com/library/antikythera-mechanism-4)is not converting...

`03:52:48  Importing: Z:/Media/antikythera/Plate Back.ipt
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 1 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced CURVE ref 10 but found spline
03:52:55  Expeced SURFACE for ref 13 but found intcurve
03:52:55  Expeced SURFACE for ref 13 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 13 but found intcurve
03:52:55  Expeced SURFACE for ref 13 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 22 but found pcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 9 but found intcurve
03:52:55  Expeced SURFACE for ref 22 but found pcurve
03:52:55  Expeced SURFACE for ref 22 but found pcurve
03:52:55  Expeced SURFACE for ref 22 but found pcurve
03:52:55  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\gjpc\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 122, in insert
    _open(filename, skip, only, root)
  File "C:\Users\gjpc\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 136, in _open
    reader.create3dModel(None , doc)
  File "C:\Users\gjpc\AppData\Roaming\FreeCAD\Mod\InventorLoader\Import_IPT.py", line 279, in create3dModel
    convertModel(root, doc.Name)
  File "C:\Users\gjpc\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 156, in convertModel
    ImportGui.insert(stepfile, docName)
<class 'RuntimeError'>: Access violation` 

[Plate Back.zip](https://github.com/jmplonka/InventorLoader/files/6148294/Plate.Back.zip)

It's a very complicated piece with spiral slots made of opposing  half circles.  Care to take a look?

jmplonka, 2021-03-16:
No - not a buzzkiller.
But only one fix a day!😄

jmplonka, 2021-03-20:
I think, it's working now:
[image]
No unresolved refs anymore ?!? Hope this fixes a lot of other issues.

gjpc, 2021-03-20:
Bravo! The load in FreeCAD looks flawless. Do you have any suggestions how I can get all the glyphs in the original file transferred into FreeCAD? I am still FreeCAD novice.

[image]

jmplonka, 2021-03-20:
Maybe it's because of the unsupported "texture" feature.
You can apply an image to a surface, but currently this is missing in the importer "View->Texture mapping..."

closed 2021-03-29 by jmplonka