jmplonka/InventorLoader #35: UnicodeDecodeError when attempt to import F3D file is made

kmarty, 2021-03-28:
Hi,
when I try to open Fusion360 (.F3D) file, I get this:
```
20:53:51  Importing: /Users/kmarty/Documents/3D tisk/tiskarna/Veci pro tiskarnu/Fan duct/Chodec/Cable_holder_E3V2_v3.f3d
20:53:51  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/kmarty/Library/Preferences/FreeCAD/Mod/InventorLoader/importerIL.py", line 122, in insert
    _open(filename, skip, only, root)
  File "/Users/kmarty/Library/Preferences/FreeCAD/Mod/InventorLoader/importerIL.py", line 127, in _open
    reader = read(filename)
  File "/Users/kmarty/Library/Preferences/FreeCAD/Mod/InventorLoader/importerIL.py", line 70, in read
    if (importerF3D.read(filename)):
  File "/Users/kmarty/Library/Preferences/FreeCAD/Mod/InventorLoader/importerF3D.py", line 94, in read
    read_manifest(f3d, name)
  File "/Users/kmarty/Library/Preferences/FreeCAD/Mod/InventorLoader/importerF3D.py", line 73, in read_manifest
    t8,  i = getLen32Text16(data, i) # UID
  File "/Users/kmarty/Library/Preferences/FreeCAD/Mod/InventorLoader/importerUtils.py", line 854, in getLen32Text16
    txt = data[i: end].decode('UTF-16LE')
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.8/encodings/utf_16_le.py", line 16, in decode
    return codecs.utf_16_le_decode(input, errors, True)
<class 'UnicodeDecodeError'>: 'utf-16-le' codec can't decode byte 0x01 in position 190: truncated data
```

FreeCAD 0.19 (24276), InventorLoader installed from Tools->Addon Manager.

luzpaz, 2021-03-28:
can you add the file?

kmarty, 2021-03-28:
(unable to attach one, I hope it is downloadable this way)
[Cable_holder_E3V2_v3.f3d@thingiverse](https://www.thingiverse.com/download:9399058)

jmplonka, 2021-03-29:
Should work a little bit better now. You can import the extracted SMB or SMBH files for your assembly.

closed 2021-03-29 by jmplonka