jmplonka/InventorLoader #36: Imported F3D object is broken

kmarty, 2021-03-29:
Hi again,
at this moment I am able to import F3D file
[image].

but either I choose "Convert to STEP" or "nativ" ("Native"?), I get this:
[image]

with "Report view" output:
```
15:32:59  Importing: /Users/kmarty/Documents/3D tisk/CAD/FreeCAD - InventorLoader/Cable_holder_E3_v113.f3d
15:32:59      ... parsing 'FusionAssetName[Active]/Breps.BlobParts/BREP.7c297a49-4f5b-46c6-b97f-c06976f48b59.smbh'
15:32:59      ... parsing 'FusionAssetName[Active]/Breps.BlobParts/BREP.dd450d24-1eaa-43db-9426-fc23dbfc27a8.smb'
15:33:13      can't create face for -1360 plane-surface $-1 -1 $-1 -28.5 11.35 0 0 0 1 1 0 0 forward_v I I I I #!
15:33:13      can't create face for -1435 plane-surface $-1 -1 $-1 3.85 10 0 -0 0 1 1 -0 0 forward_v I I I I #!
15:33:13      can't create face for -1435 plane-surface $-1 -1 $-1 3.85 10 0 -0 0 1 1 -0 0 forward_v I I I I #!
15:33:13      can't create face for -1450 plane-surface $-1 -1 $-1 -3.9 15.15 0 0 0 1 1 0 0 forward_v I I I I #!
15:33:13      can't create face for -275 plane-surface $-1 -1 $-1 26 18.8293 5.7 0 0 1 1 0 0 forward_v I I I I #!
15:33:13      can't create face for -320 plane-surface $-1 -1 $-1 -7 20 0 -1 -3.17207e-16 0 0 -0 1 forward_v I I I I #!
15:33:13      can't create face for -425 plane-surface $-1 -1 $-1 64 20 0 0 1 -0 -1 0 0 forward_v I I I I #!
```

F3D file: [Cable_holder_E3_v113.f3d](https://www.thingiverse.com/download:9390536)
or: [Cable_holder_E3V2_v3.f3d](https://www.thingiverse.com/download:9399058)
(both does the same or very similar result)

This is how it should look (this one is imported STL -> "Create shape from mesh" -> "Refine shape" -> "Convert to solid"):
[image]

BTW. Is it ok that after import a (temporary?) folder with the same name as F3D file remains next to F3D file?

jmplonka, 2021-03-31:
Hello,

the ACIS (SAT, SAB/SMB[H]) import is not easy to process for me. The result doesn't look like what is expeted and I don't understand why.

PS the folder contains the Content of the imported file for debugging purposes. Like the SMB/SMBH files for the F3D files, together with the converted SAT and / or STEP files.

kmarty, 2021-03-31:
I get it. It's just a report.

Temporary directory/folder: I see.

jmplonka, 2021-04-09:
Unfortunately I can't figure out why the plane is not well alligned. Importing the points from the resulting SAT file as point-cloud it looks the same. Nither Inventor nor Fusion360 is able to import the SMB file *LOL*.

kmarty, 2021-04-09:
So, the issue is there, it is not resolved, but closed anyway? Huh?

Btw. the original file was created in Fusion360.

jmplonka, 2021-04-09:
Please, do me a favour: have a look into the ~_f3d folder and open the ~.smbh file in FreeCAD with "Convert to STEP". Is it possible for you to open the file, or does FreeCAD crashes (QT5-Conda issue ?!?)?

kmarty, 2021-04-09:
Step by step (just for sure):
- Open FreeCAD
- update "InventorLoader" plugin:
```
Updating 9373104..dac0d12
Fast-forward
 Acis.py        | 45 ++++++++++++++++++++++++++-------------------
 importerDXF.py |  3 ---
 importerSAT.py | 32 ++++++++++++++++++++------------
 3 files changed, 46 insertions(+), 34 deletions(-)

Workbench successfully updated. Please restart FreeCAD to apply the changes.
```
- restart FreeCAD
- download "Cable_holder_E3V2_v3.f3d"
- import "Cable_holder_E3V2_v3.f3d" in FreeCAD with "Convert to STEP" (which creates temp folder with SMB and SMBH files)
- ignore how looks imported model
- close FreeCAD without saving
- open FreeCAD
- import SMBH file with "Convert to STEP"
; imported with:
```
11:27:25  Importing: /private/tmp/p/Cable_holder_E3V2_v3_f3d/BREP.6df0b3ba-94bf-46cd-8f28-160e09920178.smbh
11:27:29      can't create face for -30 cone-surface $-1 -1 $-1 78.775 110 19.25 -0 -0 -1 0.141421 -0.141421 0 1 I I 0 -1 0.2 forward I I I I #!
11:27:29      can't create face for -49 plane-surface $-1 -1 $-1 130 94.1466 28.5 0 0 1 1 0 0 forward_v I I I I #!
11:27:29      can't create face for -245 plane-surface $-1 -1 $-1 95 85 28.5 0 1 0 0 0 -1 forward_v I I I I #!
11:27:29      can't create face for -179 cone-surface $-1 -1 $-1 200 90 19.25 0 0 1 -0.0707107 -0.0707107 0 1 I I 0 1 0.1 forward I I I I #!
11:27:29      can't create face for -245 plane-surface $-1 -1 $-1 95 85 28.5 0 1 0 0 0 -1 forward_v I I I I #!
11:27:29      can't create face for -346 plane-surface $-1 -1 $-1 -35 76.0815 86 0 0 -1 -3.17207e-16 1 -0 forward_v I I I I #!
11:27:29      can't create face for -163 plane-surface $-1 -1 $-1 95 100 28.5 -1 0 0 -0 -0 -1 forward_v I I I I #!
11:27:29      can't create face for -238 plane-surface $-1 -1 $-1 95 100 10 -2.77556e-17 -0 -1 1 -0 -2.77556e-17 forward_v I I I I #!
11:27:29      can't create face for -535 plane-surface $-1 -1 $-1 160 50 38.5 0 0 1 1 0 0 forward_v I I I I #!
11:27:29      can't create face for -326 plane-surface $-1 -1 $-1 195 100 10 1 0 0 -0 -0 1 forward_v I I I I #!
11:27:29      can't create face for -231 plane-surface $-1 -1 $-1 320 100 0 0 1 -0 -1 0 0 forward_v I I I I #!
11:27:29      can't create face for -331 torus-surface $-1 -1 $-1 130 128.293 48.5 0 0 1 1.31168 -0.2 1 0 0 forward_v I I I I #!
11:27:29      can't create face for -436 cone-surface $-1 -1 $-1 130 128.293 56 0 0 -1 -0.886337 -0.608302 0 1 I I 0.209529 0.977802 1.075 forward I I I I #!
11:27:29      can't create face for -533 cone-surface $-1 -1 $-1 130 128.293 38.5 0 0 1 -0.8245 -0.565862 0 1 I I 0 1 1 forward I I I I #!
11:27:29      can't create face for -500 cone-surface $-1 -1 $-1 181.225 110 19.25 -0 -0 1 -0.141421 -0.141421 0 1 I I 0 -1 0.2 forward I I I I #!
11:27:29      can't create face for -128 cone-surface $-1 -1 $-1 130 135 38.5 0 0 1 -0.6245 -0.5 0 1 I I 0 -1 0.8 forward I I I I #!
11:27:29      can't create face for -521 plane-surface $-1 -1 $-1 130 94.1466 113.5 0 0 1 1 0 0 forward_v I I I I #!
```
in "Report view"
but it look, for first sight, fine. No crash happened in any step.

FreeCAD version 0.19 (24276) (downloaded from [FreeCAD download](https://www.freecadweb.org/downloads.php)  ) @ macOS 11.2.3

jmplonka, 2021-04-09:
So there seems to be a big difference in SMB and SMBH files and I will switch to SMBH instead! Thank you very much for your support.

jmplonka, 2021-05-24:
For a better understanding of the F3D files, I'm working on a feature based ('native') import routine for Fusion360.

closed 2021-06-29 by jmplonka