jmplonka/InventorLoader #37: Automatic install of 3rd party packages

wwmayer, 2021-04-16:
The function https://github.com/jmplonka/InventorLoader/blob/master/InitGui.py#L12 tries to install 3rd party packages if they are not available on the target system.
Note that in the call `subprocess.call(u"\"%s\" -m pip install \"%s\"" %(sys.executable, module))` the sys.executable will expand to the FreeCAD executable instead of the Python executable. Since FreeCAD doesn't support the -m option it will raise an error dialog as reported here: https://forum.freecadweb.org/viewtopic.php?f=4&t=57235

But it's in general problematic if an application tries to silently install software without asking the user for permission.

Additionally on Linux systems the preferred way is to install software with the package manager. One has to be careful to install packages via pip there because it can make the system unstable when a package has a dependency to a 3rd party library and overwrites a library on the system.

IMO, the best is to print an error message that a required package is missing and maybe a hint how it can be installed.

luzpaz, 2021-04-16:
The forum thread "[Metadata: One Standard to Rule Them All](https://forum.freecadweb.org/viewtopic.php?f=10&t=57448)" may be of interest to keep an eye on in this regard.

jmplonka, 2021-05-28:
replacing "FreeCAD" by "python" from sys.ececutable did the trick.
see Issue #40

closed 2021-05-28 by jmplonka