jmplonka/InventorLoader #38: missing requirements

mark-hahn, 2021-05-10:
When I try to use the addon manager I get this error.  I used pip to install all four python requirements.

"Some errors were found that prevent to install this workbench: Some errors were found that prevent to install this workbench. Please install the missing components first."

I used windows 10 powershell to install python 3 and pip.  I'm running freecad 0.19.

Is there a way to find out what requirement is missing?  Any ideas on how to get past this problem?

jmplonka, 2021-05-12:
Hello Mr. Hahn,
installing python will add only an additional separate python installation. FreeCAD comes with its own one. To install missing packages I suggest to follow [https://forum.freecadweb.org/viewtopic.php?f=22&t=20575&start=130#p426110](url)

sirus20x6, 2021-05-24:
clicked that link. No results matched your search.

jmplonka, 2021-05-28:
Should be fixed now.

jmplonka, 2021-06-17:
May I close this issue?

closed 2021-06-17 by mark-hahn