jmplonka/InventorLoader #40: Unrecognized option '-m'

poulet67, 2021-05-25:
Hi,

I was not able to install this mod via the add-on manager. It gave me an error message.

So I did the manual install option as documented. I get this error every time I start up

[image]

I tried the "fixing installation problems" section but it doesn't make any sense. There is no libs.zip file to unzip, or any installLibs.py file to execute (in the libs folder).

I tried opening an IPT file, but I get the following error log output:

`15:49:33  During initialization the error "No module named 'olefile'" occurred in C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\InitGui.py
15:49:33  Please look into the log file for further information
15:49:41  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 8, in <module>
    import os, sys, FreeCAD, FreeCADGui, importerSAT, importerDXF, Import_IPT, importerF3D
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 9, in <module>
    from importerUtils   import logInfo, logWarning, logError, logAlways, getUInt8A, getUInt32, chooseImportStrategyAcis, STRATEGY_SAT, setDumpFolder, getDumpFolder
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerUtils.py", line 13, in <module>
    from olefile       import OleFileIO
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'olefile'
15:50:21  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 8, in <module>
    import os, sys, FreeCAD, FreeCADGui, importerSAT, importerDXF, Import_IPT, importerF3D
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 9, in <module>
    from importerUtils   import logInfo, logWarning, logError, logAlways, getUInt8A, getUInt32, chooseImportStrategyAcis, STRATEGY_SAT, setDumpFolder, getDumpFolder
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\me\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerUtils.py", line 13, in <module>
    from olefile       import OleFileIO
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'olefile'`

hendryjl, 2021-05-26:
Same issue. I even extracted all of the .zips in the libs folder to that folder and that didn't make a difference either. Tried path and shortcut modifications as described as well.

jmplonka, 2021-05-28:
hope it is fixed now - Sorry for all the inconveniences.

closed 2021-06-29 by jmplonka