jmplonka/InventorLoader #41: codec error with some Windows locales

99RP, 2021-06-03:
Using Greek locale on windows produces codec error when attempting to import IP file.

[image]

Produces:
`['charmap' codec can't decode byte 0x9f in position 2258: character maps to <undefined>]`

jmplonka, 2021-06-08:
Hello,
I‘m sad that you‘re struggling with the encoding. Unfortunately I won‘t be able to work on this until end of next week.
I didn‘t made any restrictions on the branches – but will check it.

jmplonka, 2021-06-23:
please, could you share the example as well?

maorly, 2021-07-02:
Thank you for useful tools.
This may not relate to the thread, some environment (including Japanese cp932) required utf-8 option in opening function.
e.g.: 
with open(os.path.join(os.path.dirname(os.path.abspath(__file__)),"colors.json"),encoding="utf-8") as colorPalette:
in importerUtils.py, line 179.