jmplonka/InventorLoader #42: Python errors when importing .ipt file

crankycyclops, 2021-06-06:
Hi,

I recently encountered the below errors when trying to import an .ipt file using your code from the master branch. The shape appeared correct in the preview before the actual import. I can provide the .ipt file if necessary. Thank you for this amazing add-on!

```
14:39:30  Importing: /home/james/Desktop/potentiometer/Potentiometer_Shaft.ipt
14:39:35  ERROR> (002E): Dimension_Diameter2D - Invalid parameters: ('Diameter', 0, <Length object at 0x7f2f28101f10>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerIL.py", line 112, in insert
    reader = read(filename, False)
TypeError: read() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 2100, in Create_Sketch_Node
    addSketchObj(node, sketchObj)
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 1956, in addSketch_Dimension_Diameter2D
    constraint = Sketcher.Constraint('Diameter', index, dimension.getValue())
TypeError: Invalid parameters: ('Diameter', 0, <Length object at 0x7f2f28101f10>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  ERROR> (0034): Dimension_Diameter2D - Invalid parameters: ('Diameter', 1, <Length object at 0x7f2f28101f10>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerIL.py", line 112, in insert
    reader = read(filename, False)
TypeError: read() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 2100, in Create_Sketch_Node
    addSketchObj(node, sketchObj)
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 1956, in addSketch_Dimension_Diameter2D
    constraint = Sketcher.Constraint('Diameter', index, dimension.getValue())
TypeError: Invalid parameters: ('Diameter', 1, <Length object at 0x7f2f28101f10>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Do not use this function directly; instead, use 'make_ortho_array', 'make_polar_array', or 'make_circular_array'.
14:39:35  <Exception> type must be int, not float
14:39:35  ERROR> (00DC): Dimension_Diameter2D - Invalid parameters: ('Diameter', 0, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerIL.py", line 112, in insert
    reader = read(filename, False)
TypeError: read() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 2100, in Create_Sketch_Node
    addSketchObj(node, sketchObj)
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 1956, in addSketch_Dimension_Diameter2D
    constraint = Sketcher.Constraint('Diameter', index, dimension.getValue())
TypeError: Invalid parameters: ('Diameter', 0, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  ERROR> (00E2): Dimension_Diameter2D - Invalid parameters: ('Diameter', 1, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerIL.py", line 112, in insert
    reader = read(filename, False)
TypeError: read() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 2100, in Create_Sketch_Node
    addSketchObj(node, sketchObj)
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 1956, in addSketch_Dimension_Diameter2D
    constraint = Sketcher.Constraint('Diameter', index, dimension.getValue())
TypeError: Invalid parameters: ('Diameter', 1, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  ERROR> (0153): Dimension_Diameter2D - Invalid parameters: ('Diameter', 1, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerIL.py", line 112, in insert
    reader = read(filename, False)
TypeError: read() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 2100, in Create_Sketch_Node
    addSketchObj(node, sketchObj)
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 1956, in addSketch_Dimension_Diameter2D
    constraint = Sketcher.Constraint('Diameter', index, dimension.getValue())
TypeError: Invalid parameters: ('Diameter', 1, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  ERROR> (017D): Dimension_Diameter2D - Invalid parameters: ('Diameter', 0, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index

14:39:35  Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerIL.py", line 112, in insert
    reader = read(filename, False)
TypeError: read() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 2100, in Create_Sketch_Node
    addSketchObj(node, sketchObj)
  File "/home/james/.FreeCAD/Mod/InventorLoader/importerFreeCAD.py", line 1956, in addSketch_Dimension_Diameter2D
    constraint = Sketcher.Constraint('Diameter', index, dimension.getValue())
TypeError: Invalid parameters: ('Diameter', 0, <Length object at 0x7f2f28101ee0>)
Constraint constructor accepts:
-- empty parameter list
-- Constraint type and index
```

jmplonka, 2021-06-08:
Hi,
Sorry for the inconveniences. Could you please send me the file, so I can fix this issue.

crankycyclops, 2021-06-08:
Hi @jmplonka, no worries. I really appreciate your work on this. It's incredibly useful :) Here's a link to a ZIP archive containing the affected .ipt files: [blog.jeffcolemanwrites.com/downloads/potentiometer.zip](url)

luzpaz, 2023-04-02:
@jmplonka is there anything left for this ticket ?