jmplonka/InventorLoader #46: Errors With Inventor Parts

infosisio, 2021-06-18:
This visually imports fine but has a lot of errors in the console:
https://grabcad.com/library/studycadcam-3d-45-cad-exercise-44-autodesk-inventor-pro-1

This one has a lot of errors in the console and has no holes:
https://grabcad.com/library/studycadcam-3d-cad-exercise-100-autodesk-inventor-pro-1

jmplonka, 2021-06-23:
Thank you for sharing the files. Let's see what I can do for you.

jmplonka, 2021-06-24:
Please, can you send me a feedback about the chagnes?

infosisio, 2021-06-25:
https://grabcad.com/library/studycadcam-3d-cad-exercise-100-autodesk-inventor-pro-1 -> Still no holes.

https://grabcad.com/library/studycadcam-3d-45-cad-exercise-44-autodesk-inventor-pro-1 -> Correct.

https://grabcad.com/library/practice-part-9 -> No holes.

marcocecchiscmgroup, 2023-04-26:
Should be fixed by 372beb8c1290fdd9950642ddbe793cb2ceee8ca8