jmplonka/InventorLoader #47: Some errors were found that prevent to install this workbench: Some errors were found that prevent to install this workbench. Please install the missing components first.

poulet67, 2021-06-30:
I keep getting this error on a fresh Freecad Install (0.19) on Windows 10 when trying to install from the Addon Manager.

jmplonka, 2021-07-06:
Please, can you please send me the text in the output window, because I've no Idea about what is going wrong.
What OS (Linux, OS-X, Windows) are you using?

MariwanJ, 2021-07-25:
Some errors were found that prevent to install this workbench: Some errors were found that prevent to install this workbench. Please install the missing components first.

luzpaz, 2021-07-25:
@MariwanJ TBF current HEAD is at 25310 and you're using 24840

MariwanJ, 2021-07-25:
Ohh! . so long time I was only focusing on my learning/making the new tools :).
I will update my freecad.

eflukx, 2022-03-28:
Same issue here. Was this resolved in the mean time?
No clue what dependencies might be missing. Need to dig up a proper log somewhere, the output in the FreeCAD GUI isn't of much help...

luzpaz, 2022-03-28:
The FreeCAD Addon Manger has been updated in v0.20 significantly. Please install the latest (nightly) development version and test installing this addon "InventorLoader" to see if the Addon Manager does indeed install all dependencies accordingly

poulet67, 2022-04-06:
It works great now! Thank you so much!

closed 2022-04-06 by poulet67