jmplonka/InventorLoader #49: Dxfgrabber obsolete

marcocecchiscmgroup, 2021-08-30:
Hello Jens,
as you might remember, we had an exchange about a bunch of new features that I was suggesting in your InventorLoader project a few years ago. I've seen that you made more progress with it and that sounds great. The support for F3D is very appreciated in this respect.
I did a sync lately, so as to update my own project, and I have noticed that you now include dxfgrabber to allow 3d import from dxf, exactly as we discussed at that time. I just wanted to let you know, if you are ok with it, that dxfgrabber is obsolete and does not support entities such as all the four surfaces (revolved, swept, ...), that are as well of type BODY. You should refer to ezdxf, instead, lest supporting only a tiny subset of all the 3D entities.
From the doc: "ezdxf is also a replacement for my dxfwrite and my dxfgrabber packages but with different APIs [...] Since ezdxf can do all the things that dxfwrite and dxfgrabber can do, I focused on the development of ezdxf, dxfwrite
and dxfgrabber are in maintenance mode only and will not get any new features, just bugfixes."

Keep up with the good work buddy!

marcocecchiscmgroup, 2021-10-20:
Kudos again for the latest features, such as support to Fusion 360.
I can help, in case.

marcocecchiscmgroup, 2023-04-27:
Integrating ezdxf is one line of code and will allow to import the four kind of surfaces that are foreseen in the dxf format.: LOFTED, SWEPT, EXTRUDED and REVOLVED.

jmplonka, 2023-04-28:
done.

jmplonka, 2023-04-29:
By replacing dxfgrabber by ezdxf other "bodies" can now  also be imported.