jmplonka/InventorLoader #51: encoding and decoding

superbin1996, 2021-11-22:
Dear sir,
In those below files, maybe more due to my weak memory.
importerUtils.py, InventorViewProviders.py, importerIL.py, Import_IPT.py, importerReader.py

instead of <decoding = 'utf-8'> and <encoding = 'utf-8'>, 
you typed <'utf-8'> and <encoding = 'utf8'> and <decoding = 'utf8'>

Normal people with English windows don't have this problem. But i am using Japanese windows, so it can't run and there are a lot of  error alert. 
I can see many people have this problem too. 

It is simple too fix anyway if the alert were more accurate

jmplonka, 2021-12-26:
Hello,

sorry for the inconveniences. Indeed I had lots of problems to get it working on my machine.

Until now I'm facing some issues:
a) Missing sample files to reproduce the behaviour.
b) No idear how to change coding (e.g. japanese) without loosing information about what I see / do. 

Maybe you can give me some advice how come along?

Sincerely yours
Jens M. Plonka

dinosauria123, 2022-01-20:
This problem may fix by attached file.
[importerUtils.py.txt](https://github.com/jmplonka/InventorLoader/files/7907731/importerUtils.py.txt)

jmplonka, 2022-01-21:
Fix is uploaded.

closed 2022-04-26 by jmplonka