jmplonka/InventorLoader #52: Inventor .iam import

marcin86junior, 2021-11-29:
Hello! 
I have question regarding .iam - any chance to get import of iam and 2 parts? How to decode .iam?
Thank for you all work :) 
Best regards
Marcin Nowak

jmplonka, 2021-12-26:
Hello,

untill now I can't say much about IAM files. The IAM files consist - like IPT - of segments. If you want to deciver IAM I suggest to reverse engineer your work. Like add a part. Move it, save, Rotate, save, ... Each step with a different file name.
The reader is nearly prepared to dump the IAM content in a apropriate subdirectory as as a readable Textfile.
Than you can diff the steps and see what changes you got for each step.

Best regards, and Marry Christmas
Jens Plonka
The read

Mannshoch, 2024-12-13:
Any news on this case?

jmplonka, 2024-12-13:
Unfortunately there are only links to IPT‘s. 
I don’t think I will implement this - Sorry.