jmplonka/InventorLoader #53: Loading sat file from IdealStandard: Acis.py", could not convert string to float: 'F'

varnierg, 2022-01-09:
I'm using the latest version of InventorLoader (22 oct 2021) to load a .sat files of a furniture downloaded from Idealstandard ([Link](https://www.idealstandard.it/prodotti/catalog/bidets/bidet/t5150#downloads)), that I've attached.
When I try to load it in freecad 0.19, I get this error (below complete error message):

Acis.py", line 335, in getFloat    return float(val), in <class 'ValueError'>: could not convert string to float: 'F'

I've the same error if I import it as native or convert to STEP.

Bye and thank you!
[IS_Ventuno_T5150_3DPL_NN.zip](https://github.com/jmplonka/InventorLoader/files/7834942/IS_Ventuno_T5150_3DPL_NN.zip)

Full error

```
12:23:35  Reading: C:\Users\andrea\Downloads\IS_Ventuno_T5150_3DPL_NN.SAT
12:23:37  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 147, in open
    _open(filename, skip, only, root)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 135, in _open
    reader.create3dModel(root , doc)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 203, in create3dModel
    importModel(group)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 164, in importModel
    bodies = resolveNodes(acis)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 145, in resolveNodes
    node = createNode(entity)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 271, in createNode
    node.set(entity)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 2646, in set
    i = self.setSubtype(entity.chunks, i)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 3657, in setSubtype
    i = self.setBulk(chunks, i + 1)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 3651, in setBulk
    return fkt(chunks, i + prm[1], prm[2])
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 3180, in setExact
    rU, i = getInterval(chunks, i, MIN_INF, MAX_INF, getScale())
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 436, in getInterval
    upper, i = getRange(chunks, i, defMax, scale)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 428, in getRange
    val, i = getFloat(chunks, i)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 335, in getFloat
    return float(val), i
<class 'ValueError'>: could not convert string to float: 'F'
12:30:56  Reading: C:\Users\andrea\Downloads\IS_Ventuno_T5150_3DPL_NN.SAT
12:30:58  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 147, in open
    _open(filename, skip, only, root)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerIL.py", line 135, in _open
    reader.create3dModel(root , doc)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 205, in create3dModel
    convertModel(group, doc.Name)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 171, in convertModel
    bodies = resolveNodes(acis)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\importerSAT.py", line 145, in resolveNodes
    node = createNode(entity)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 271, in createNode
    node.set(entity)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 2646, in set
    i = self.setSubtype(entity.chunks, i)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 3657, in setSubtype
    i = self.setBulk(chunks, i + 1)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 3651, in setBulk
    return fkt(chunks, i + prm[1], prm[2])
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 3180, in setExact
    rU, i = getInterval(chunks, i, MIN_INF, MAX_INF, getScale())
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 436, in getInterval
    upper, i = getRange(chunks, i, defMax, scale)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 428, in getRange
    val, i = getFloat(chunks, i)
  File "C:\Users\andrea\AppData\Roaming\FreeCAD\Mod\InventorLoader\Acis.py", line 335, in getFloat
    return float(val), i
<class 'ValueError'>: could not convert string to float: 'F'
```

jmplonka, 2022-01-21:
Thank you for sharing the example. Can you please check if it is working now?

varnierg, 2022-01-21:
yes, it's working, thanks!

closed 2022-01-21 by jmplonka