jmplonka/InventorLoader #54: Update to new FreeCAD AddonManager metadata schema

luzpaz, 2022-01-27:
https://github.com/FreeCAD/FreeCAD-addons/issues/200

luzpaz, 2022-02-15:
@jmplonka ping

luzpaz, 2022-02-15:
https://github.com/jmplonka/InventorLoader/blob/master/metadata.txt needs to be migrated to `package.xml` and modified

jmplonka, 2022-02-15:
Done - plaese, check if valid.

luzpaz, 2022-02-15:
@jmplonka in the meanwhile the `metadata.txt` file can be removed :wink:

chennes, 2022-02-15:
@luzpaz actually that's not true, you want to keep that around for the 
```
[dependencies]
workbenches=Part,Draft,Sketcher
pylibs=olefile,xlrd,xlwt,xlutils
FCaddon=sheetmetal
```

The rest of it is ignored, but the Python deps are still read from either `requirements.txt` or `metadata.txt`. And yes, the Addon Manager will offer to install the missing dependencies, as long as they are on the approved list. I'll look at adding these.

luzpaz, 2022-02-16:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/AddonManager/ALLOWED_PYTHON_PACKAGES.txt FTW!

chennes, 2022-02-17:
Oh yeah, right! This WB was one of my test cases, of course the deps already work. I totally forgot.

luzpaz, 2022-02-17:
Fixed in #55  
~@chennes having an issue loading the InventorLoader in the addon manager. It's looking for a `main` branch when the default is `master`~
[image]

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27571 (Git)
Build type: Release
Branch: master
Hash: 0bbc253ddee6801770c78e011c5201b7c2c36080
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * ToolbarStyle
  * stickframe
  * sheetmetal 0.2.49
  * Behave-Dark-Colors 0.0.1
  * Curves 0.3.0
  * Ship
  * BIM 2021.12.0
```

luzpaz, 2022-04-14:
Can this be closed?

jmplonka, 2022-04-26:
Who can tell? Yes from my side :)

luzpaz, 2022-04-26:
@chennes can we close this?

chennes, 2022-04-26:
Yes, InventorLoader's metadata seems fine, it displays and installs without issue, including its stated dependencies.

closed 2022-04-26 by luzpaz