jmplonka/InventorLoader #56: So many errors

iBlz, 2022-05-22:
18:49:43  Reading: C:\Users\skepp\Downloads\User Library-D-Link 1Gb Switch 16-Port (2)\User Library-D-Link 1Gb Switch 16-Port.SAT
18:49:45  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\importerIL.py", line 147, in open
    _open(filename, skip, only, root)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\importerIL.py", line 135, in _open
    reader.create3dModel(root , doc)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\importerSAT.py", line 183, in create3dModel
    convertModel(group, doc.Name)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\importerSAT.py", line 149, in convertModel
    bodies = resolveNodes(acis)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\importerSAT.py", line 127, in resolveNodes
    node = createNode(entity)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\Acis.py", line 263, in createNode
    node.set(entity)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\Acis.py", line 3971, in set
    self.value, i = getInteger(entity.chunks, i)
  File "C:\Users\skepp\AppData\Local\Programs\FreeCAD 0.19\Mod\InventorLoader-master\Acis.py", line 293, in getInteger
    return int(val), i
<class 'ValueError'>: invalid literal for int() with base 10: '1Gb'

luzpaz, 2022-05-25:
Please enclose in code tags.

luzpaz, 2023-04-02:
@iBlz :point_up:

HolographicPrince, 2023-04-15:
Send this file

jmplonka, 2023-04-29:
@iBlz : Any test file to share?