jmplonka/InventorLoader #59: FreeCad crashes trying to import this file

bitingsock, 2022-08-28:
Installed from the Addon Manager.
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * InventorLoader 1.3.0
  * sheetmetal 0.2.56
```

[crash.log](https://github.com/jmplonka/InventorLoader/files/9439415/crash.log)
[Fully parametric ball bearing.zip](https://github.com/jmplonka/InventorLoader/files/9439416/Fully.parametric.ball.bearing.zip)

jmplonka, 2022-10-25:
It's because of an access violation caused by trying to generate a cone surface out of a revoluted line.
eg:
`import Part
V = App.Vector
apex   = V(0.0, 0.0, -65.00000000000001)
center = V(0.0, 0.0, -26.25)
major  = V(38.75, 0.0, 0.0)
axis   = V(0.0, 0.0, -1.0)
line   = Part.Line(apex, center + major).toShape()
line.revolve(center, axis, 360.0)`

jmplonka, 2022-10-25:
According to Chris_G, line has to be created not with Part.Line but rather with Part.LineSegment.

luzpaz, 2022-10-26:
Can freecad python interpreter be adjusted to recognize this an output recommended command instead of crashing ?

numericOverflow, 2022-11-07:
I've got a similar issue. FreeCad 0.20.1 crashes whenever I try to import the attached *.f3d file.

[crash.log](https://github.com/jmplonka/InventorLoader/files/9947141/crash.log)

[f3d_files.zip](https://github.com/jmplonka/InventorLoader/files/9947148/f3d_files.zip)
The "Holder_v13.f3d" crashes freecad
"Front_Rev_1_v9.f3d" doesn't crash freecad, but nothing gets converted.

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Dracula 0.0.2
  * InventorLoader 1.3.0

luzpaz, 2023-04-02:
@tomate44 do you mind weighing in again ?

albfan, 2025-04-23:
import using convert to STEP works. (inventorLoader 1.5.0, freecad 1.0 appimage)

[image]

So probably this is more about avoid a crash on nativ and a way to log errors in that mode