jmplonka/InventorLoader #61: Crashes importing *.f3d

mfraser, 2022-12-29:
FreeCAD 0.21 crashes when trying to import shroud.f3d from https://github.com/spuder/ender2pro-raspberrypi/tree/main/CAD.
Am I just expecting too much from this plugin?

luzpaz, 2022-12-29:
Error message?  
[About](https://wiki.freecad.org/About) info ?

mfraser, 2022-12-29:
[code]
OS: Ubuntu 22.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31432 (Git) AppImage
Build type: Release
Branch: master
Hash: 63a2855751d638bf27ed969dc60654bda2b54baf
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.2.60
  * freecad.gears 1.0.0
  * Manipulator 1.4.9
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Lithophane
  * Curves 0.6.5
  * Assembly4 0.12.4
  * fasteners.backup1668342244.1881278 (Disabled)
  * InventorLoader 1.3.0
  * fasteners 0.4.50
  * A2plus 0.4.60g
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.84.0
  * Assembly3 0.11.4
  * VendorParts
[/code]

Importing: /home/mfraser/Downloads/Shroud.f3d
    ... parsing 'FusionAssetName[Active]/Breps.BlobParts/BREP.0dcc3cb6-9f92-41ce-b2d8-376b45310ca5.smbh'
    ... parsing 'FusionAssetName[Active]/Breps.BlobParts/BREP.32b93ae7-9fa6-434e-942f-ec983edc9bc9.smb'
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f7d825cf520]
#1  0x7f7d621a784d in BRepSweep_Rotation::IsInvariant(TopoDS_Shape const&) const from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0x2d
#2  0x7f7d621a7960 in BRepSweep_Rotation::IsInvariant(TopoDS_Shape const&) const from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0x140
#3  0x7f7d621abc43 in BRepSweep_Trsf::Process(TopoDS_Shape const&, Sweep_NumShape const&) from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0x73
#4  0x7f7d621ac11f in BRepSweep_Trsf::Init() from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0x7f
#5  0x7f7d621a217a in BRepSweep_Rotation::BRepSweep_Rotation(TopoDS_Shape const&, Sweep_NumShape const&, TopLoc_Location const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0x9a
#6  0x7f7d621a1bff in BRepSweep_Revol::BRepSweep_Revol(TopoDS_Shape const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0xdf
#7  0x7f7d621bbd2c in BRepPrimAPI_MakeRevol::BRepPrimAPI_MakeRevol(TopoDS_Shape const&, gp_Ax1 const&, double, bool) from /tmp/.mount_FreeCAAtyZWT/usr/lib/./libTKPrim.so.7+0x6c
#8  0x7f7d64f03735 in Part::TopoShape::revolve(gp_Ax1 const&, double, bool) const from /tmp/.mount_FreeCAAtyZWT/usr/lib/Part.so+0xe5
#9  0x7f7d64dbd9c0 in Part::TopoShapePy::revolve(_object*) from /tmp/.mount_FreeCAAtyZWT/usr/lib/Part.so+0x220
#10  0x7f7d64dbdd2f in Part::TopoShapePy::staticCallback_revolve(_object*, _object*) from /tmp/.mount_FreeCAAtyZWT/usr/lib/Part.so+0x1f
#11  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(+0x1480d2) [0x7f7d840c30d2]
#12  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x2bb) [0x7f7d840bbfdb]
#13  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x54e4) [0x7f7d840b7b04]
#14  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#15  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x70a) [0x7f7d840b2d2a]
#16  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#17  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x70a) [0x7f7d840b2d2a]
#18  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#19  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#20  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#21  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#22  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#23  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#24  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#25  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#26  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#27  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#28  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#29  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#30  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#31  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x315) [0x7f7d840b2935]
#32  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#33  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x49c3) [0x7f7d840b6fe3]
#34  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x78) [0x7f7d840c3528]
#35  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x49c3) [0x7f7d840b6fe3]
#36  /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libpython3.10.so.1.0(+0x1ed6a2) [0x7f7d841686a2]
[... cut ...]
#72  0x7f7d82b7d3b5 in QCoreApplication::exec() from /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libQt5Core.so.5+0x107
#73  0x7f7d84756671 in Gui::Application::runApplication() from /tmp/.mount_FreeCAAtyZWT/usr/bin/../lib/libFreeCADGui.so+0x1ba1
#74  /tmp/.mount_FreeCAAtyZWT/usr/bin/freecad(+0x68e5) [0x5601c30598e5]
#75  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f7d825b6d90]
#76  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f7d825b6e40]
#77  /tmp/.mount_FreeCAAtyZWT/usr/bin/freecad(+0x6c29) [0x5601c3059c29]

luzpaz, 2022-12-29:
please put these in markdown code tags!
```
code tags
``` 
(the output is polluting other open issues)

mathieunicolas, 2023-01-11:
Hi, same here, when I try to open f3d files FreeCAD just crashes. I can't see anything since FreeCAD crashes instantly, but if you explain how I can get some logs I will sure provide them to you !

jmplonka, 2023-04-29:
Same with me on macOS, but not with Windows:
[code]
OS: macOS 13.3
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * SearchBar 1.0.1
  * InventorLoader 1.4.0
  * fcgear 1.0.0
  * Importer3D
  * CurvedShapes 1.0.4
[/code]

jmplonka, 2023-04-30:
See #59.