jmplonka/InventorLoader #63: Error loading DSM's sab files

radovanma, 2023-01-11:
I'm trying to import the sab files contained in RSDOC [Design Spark Mechanical](https://www.rs-online.com/designspark/mechanical-software) (v4 or V5) documents using the InventorLoader plugin (as native or convert to STEP), but some errors appear and the sab file won't open.
I tested this on FreeCad versions 0.20.2 and 0.19
You can "mine" the sab file by renaming the .rsdoc file to a .zip file (TestFile.zip\SpaceClaim\Geometry\part1bodies.sab).

Here is the FreeCad Log:
`14:04:32  Importing: C:/3Dparts/part1bodies.sab
14:04:33      Missing class implementation for '6NURTBZPF83TKQS3_EZRHXQCF6RKWDQMNC4MR27539N3QRPAFCNUSC6FBCA4R4Q3UARKFXPCBCPMJ-body%    Missing class implementation for 'string_attrib%    Missing class implementation for 'lump%    Missing class implementation for 'shell%    Missing class implementation for 'face%	    Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for 'loop%
14:04:33      Missing class implementation for 'plane%    Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for 'coedge%
14:04:33      Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for 'edge%    Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for '%
14:04:33      Missing class implementation for 'cone%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for 'pcurve%    Missing class implementation for '%    Missing class implementation for 'vertex%    Missing class implementation for '%    Missing class implementation for 'straight%    Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for 'ellipse%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '
[TestFile.zip](https://github.com/jmplonka/InventorLoader/files/10392657/TestFile.zip)
%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for 'point%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%	    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%
14:04:33      Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%    Missing class implementation for '%`

[TestFile.zip](https://github.com/jmplonka/InventorLoader/files/10392716/TestFile.zip)

marcocecchiscmgroup, 2023-04-17:
The sab version is 28, which is not supported by this plugin

jmplonka, 2023-04-24:
Hello,
Ansys hat incorporated a 'strange' feature when saving the ACIS information in binary files.
normally it is "body ... #" here we have "body%UINT32 ..." and I don't understand the meaning of the 32Bit integer value.
-Jens

jmplonka, 2023-04-24:
I think I found a way to read the format - please check out latest version.