jmplonka/InventorLoader #65: Ever considered making acis2step functionality as a standalone python module/package?

Krande, 2023-01-20:
Hey, 

first of all thank you for the awesome job you've done on this workbench! 

A while back I started writing my own SAT ACIS parser from scratch in python in [adapy](https://github.com/Krande/adapy), but I did not get very far (I think last time I stopped at 2d shells) given my lack of experience with the standard and lack of time. Then yesterday I come across your repo and see that you have done a heck of job on parsing SAT ACIS files already :) 

I was just wondering if there's any chance you would be interested in separating the acis and acis2step code into a standalone python module/package independent of any external (Freecad internals etc..) libraries so that it's possible to use it outside of Freecad? If you want I will help out in any way I can :) 

If you're not interested, then I completely understand :) Then I will just try to write my own version based on the wonderful work you've done!

Best Regards
Kristoffer

jmplonka, 2023-04-23:
Hello Kristoffer,
the idea is fascinating and the advantages are obvious.
Let me have a deeper look into the converter. But currently I think - without having a look into the code - the converter requires some of the FreeCAD libraries.
Let‘s see what I can tell you after having a look into it.
Best Regards
Jens

Krande, 2023-04-25:
Sounds good!

Let me know if there's anything I can do to help :)

Best Regards
Kristoffer

marcocecchiscmgroup, 2023-04-26:
And why such shortcomings should be ok for a FreeCAD plugin instead?

marcocecchiscmgroup, 2023-04-26:
Hint:
exploting the exportStep() primitive from Part makes the Acis2tep.py module totally unnecessary, also making up for all the missing bits&pieces that you mentioned.

Krande, 2023-04-26:
Hi Jens!

Okay, so if I understand correctly, the subtypes you are referring to are spline types that the InventorLoader manages to successfully parse from ACIS and convert to STEP as of today?

Then the functionality we need to convert the types to STEP is located somewhere inside FreeCAD? 

And if I understand @marcocecchiscmgroup correctly, then the Part class which contains the necessary functionality in Freecad is using opencascade. 

If you point me in the direction of where you pass the spline subtypes to FreeCAD in your plugin then I (or feel free to do it yourself) can create a set of "Interface classes" representing the base Spline definitions. I am assuming that the different spline representations are passed to FreeCAD somehow. And if we find the appropriate interface location we can create a set of adapters (or whatever you would call them :) ). 

And for a standalone acis2step module parsing to STEP I can write the necessary opencascade code based on how Freecad does it internally today. 

This way we can also write unittests for the data collected in the "Interface classes". 

That way we dont have to break the existing way you pass geometry information to FreeCAD and we can use the module outside of freecad. 

How does that sound? 

(I wrote this in a hurry, so please feel free to ask if I anything is unclear).

Best Regards
Kristoffer

marcocecchiscmgroup, 2023-04-26:
Hello Krande,
let me chime in again, as Jens' response times are quite unpredictable :-)

> If you point me in the direction of where you pass the spline subtypes to FreeCAD in your plugin then I (or feel free to do it yourself) can create a set of "Interface classes" representing the base Spline definitions

This project from Jens already uses Part throughout in all the acis.py module, so what you suggest is *already* in place. You just have to do a shape.exportStep(stepfilefullpath) for each body. But there's a catch: the generalFuse algorithm (from OCCT) that Jens is using to reconstruct the faces is a *wrong* approach, which is why STEP conversion via acis2step gives oftrentimes better results that the native import (in that case, you just specify the surfaces and the co-edges 1-1 with the ACIS representation).

The question is: [Part](https://wiki.freecad.org/Part_Module) is a FreeCAD module, yet based on OpenCascade, so you still would end up with all the tons of FreeCad dependencies.
If you want a standalone project, please refer to [InventorLoader-DBG](https://github.com/jmplonka/InventorLoader-DBG) where Part and FreeCAD are (partly) stubbed. However, according to Jens reply, I am afraid several spline types would be missing? How much work would be to translate them to STEP in this latter case?

jmplonka, 2023-04-28:
I reduced the dependencies a little bit, so the stubs don't need too much functionality. This is all I can do for you.

closed 2023-04-28 by jmplonka