jmplonka/InventorLoader #66: Wrong import

dronov-dmitry, 2023-04-02:
When i import this file:
https://disk.yandex.ru/d/cayCPNwgRm7w8A
by this plugin
It ask me convert to STEP and after that is wrong imported
Not like this
https://disk.yandex.ru/i/vXHWlYq3zwgfvQ
But like this
https://disk.yandex.ru/i/i4OKv9Xpj_0WJw

marcocecchiscmgroup, 2023-04-13:
The reason is because the two smaller elements, the doors, have a non trivial transform from the ACIS, but that is not applied in the STEP conversion.:

def _convertShell(acisShell, representation, shape, parentColor, transformation):
	# FIXME how to distinguish between open or closed shell?
	faces = acisShell.getFaces()
	if (len(faces) > 0):
		color = getColor(acisShell)
		defColor = parentColor if (color is None) else color
		shell = OPEN_SHELL('',[])
		for acisFace in faces:
			faces = _convertFace(acisFace, representation, defColor, representation.context)
			shell.faces += faces

			assignColor(defColor, shell, representation.context)
		return shell

	return None

jmplonka, 2023-04-23:
Hello,

Thank you very much for the detailed explanation.

Best regards
Jens

marcocecchiscmgroup, 2023-04-26:
Jens, I think I understand why you didn't initially provide an implementation for transformations in STEP, as it is a real mess.
I have a drafted an implementation that works for me based on:
ITEM_DEFINED_TRANSFORMATION, REPRESENTATION_RELATIONSHIP_WITH_TRANSFORMATION, SHAPE_REPRESENTATION_RELATIONSHIP,
CONTEXT_DEPENDENT_SHAPE_REPRESENTATION and the like

marcocecchiscmgroup, 2023-04-26:
That requires factoring out many arguments that you built implicitly.
Just check this out
[Acis2Step.zip](https://github.com/jmplonka/InventorLoader/files/11332783/Acis2Step.zip)

marcocecchiscmgroup, 2023-04-27:
As usual, I am using the -dbg stubs, so I am not sure that the official FreeCAD/Parts module implement Vector.rotate(), Rotation.axis() etc. Just in case;

```
class Rotation(object):
[...]
	def angle(self):
		return acos(self.w) * 2.0
	def axis(self):
		sin_theta_half = sin(self.angle() / 2.0)
		return Vector(self.x * sin_theta_half, self.y * sin_theta_half, self.z * sin_theta_half)
```

```
class Vector(object):
[...]
	def rotation_matrix(self, axis, angle):
		axis.normalize()
		a = cos(angle / 2.0)
		aa = a * a
		v = axis.reversed() * sin(angle / 2.0)
		b = v.x
		c = v.y
		d = v.z
		aa, bb, cc, dd = a * a, b * b, c * c, d * d
		bc, ad, ac, ab, bd, cd = b * c, a * d, a * c, a * b, b * d, c * d
		m11 = aa + bb - cc - dd
		m12 = 2 * (bc + ad)
		m13 = 2 * (bd - ac)
		m21 = 2 * (bc - ad)
		m22 = aa + cc - bb - dd
		m23 = 2 * (cd + ab)
		m31 = 2 * (bd + ac)
		m32 = 2 * (cd - ab)
		m33 = aa + dd - bb - cc
		return Matrix(m11, m12, m13, 0.0, m21, m22, m23, 0.0, m31, m32, m33, 0.0, 0.0, 0.0, 0.0, 0.0)
	def rotate(self, v, angle):
		return self.rotation_matrix(v, angle) * self
	def projectToLine(self, base, line):
		self = base * line / line.square_modulus() * line - base;
		return self

```