jmplonka/InventorLoader #67: Cannot import anymore because of  'Shape' object has no attribute 'reverse error

marcocecchiscmgroup, 2023-04-06:
Dear Jens,

starting from commit 2d4f295 everything seems broken. What is this reverse() method you introduced?
Thanks, Marco.

Reading: D:\jplonka\Demo-Status\Demo-Status-0.2.ipt
Traceback (most recent call last):
  File "D:\jplonka\_main.py", line 28, in importFolder
    importerIL.open(filename)
  File "D:\jplonka\importerIL.py", line 147, in open
    _open(filename, skip, only, root)
  File "D:\jplonka\importerIL.py", line 135, in _open
    reader.create3dModel(root , doc)
  File "D:\jplonka\Import_IPT.py", line 279, in create3dModel
    convertModel(root, doc.Name)
  File "D:\jplonka\importerSAT.py", line 151, in convertModel
    stepfile = export(acis.name, acis.header, bodies)
  File "D:\jplonka\Acis2Step.py", line 1465, in export
    bodies += _convertBody(body, appPrtDef)
  File "D:\jplonka\Acis2Step.py", line 565, in _convertBody
    shape = _convertLump(acisLump, name, appPrtDef.application, getColor(acisBody), transformation)
  File "D:\jplonka\Acis2Step.py", line 544, in _convertLump
    shell = _convertShell(acisShell, shapeRepresentation, shape, defColor, transformation)
  File "D:\jplonka\Acis2Step.py", line 520, in _convertShell
    faces = _convertFace(acisFace, representation, defColor, representation.context)
  File "D:\jplonka\Acis2Step.py", line 502, in _convertFace
    faces = _createSurface(acisFace)
  File "D:\jplonka\Acis2Step.py", line 489, in _createSurface
    shape = acisFace.build() # will Return Face!
  File "D:\jplonka\Acis.py", line 1692, in build
    surface.reverse()
AttributeError: 'Shape' object has no attribute 'reverse

luzpaz, 2023-04-11:
Not sure what is up with @jmplonka (hasn't been active on github since June 2022)

luzpaz, 2023-04-26:
@jmplonka please advise :point_up_2:

HolographicPrince, 2023-04-28:
I noticed that Jens kinrdly added some methods to the -dgb repository. You can close this ticket.

closed 2023-04-29 by jmplonka