jmplonka/InventorLoader #77: Python errors when importing .ipt file

coelhoDj, 2023-07-26:
Couldn't open an inventor2019 ipt file.
Don't know what's wrong  with my Freecad version.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Portugal (pt_PT)
Installed mods: 
  * InventorLoader 1.4.0
[/code]

This was what returned me:

"
22:42:30  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\HC\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 8, in <module>
    import os, sys, FreeCAD, FreeCADGui, importerSAT, importerDXF, Import_IPT, importerF3D
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\HC\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerDXF.py", line 8, in <module>
    from ezdxf         import readfile
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'ModuleNotFoundError'>: No module named 'ezdxf'

"
Would be grateful for your help.
Thank you for your time.

jmplonka, 2023-08-09:
Hello, would it be posssible for you to share the IPT file?

jmplonka, 2023-08-09:
I don't know, why package ezdxf is not installed?!?
As a Workaround: please open the command line tool.
Navigate to the bin path of your FreeCAD installatin (e.g <CODE>C:\Program Files\FreeCAD 0.20\bin</CODE>)
type:
#><CODE>python.exe -m pip install ezdxf</CODE>
Reopen FreeCAD and try again to import your IPT file.

gmartincan, 2023-08-09:
> Navigate to the bin path of your FreeCAD installatin (e.g C:\Program

Hi,
Maybe the problem is on inventor file version as you suggest. I've tried
other files and were imported correctly.
Attached one of the files i can't import.

Thanks

coelhoDj, 2023-08-10:
> I don't know, why package ezdxf is not installed?!? As a Workaround: please open the command line tool. Navigate to the bin path of your FreeCAD installatin (e.g `C:\Program Files\FreeCAD 0.20\bin`) type: C:\Program Files\FreeCAD 0.20\bin>`python.exe -m pip install ezdxf` Reopen FreeCAD and try again to import your IPT file.

It Worked !
I really aprecciated your help.
Have a nice day!
[image]

jmplonka, 2023-08-11:
for those who are using a Mac:
open terminal and navigate to FreeCad.app/Contents/Resources/bin (in my case:
% <CODE>cd /Applications/FreeCAD.app/Contents/Resources/bin</CODE>)
and install ezdxf:
% <CODE>./python.exe -m pip install ezdx</CODE> (don't forget <CODE>./</CODE>!)