jmplonka/InventorLoader #79: exact_int_curve is not handled?

mconsidine, 2023-09-16:
Hi,
I'm trying to debug the processing of a F3D-based shape and I've found a curve type that doesn't seem to be handled.  Thought "law"  is a case, but 'exact_int_cur' doesn't seem to be.

Is that correct?

It appears to have this type, shape, sense and spline info:
12:15:23  <class 'Acis.CurveInt'>
12:15:23  <Edge object at 0x563e36a32ca0>
12:15:23  forward
12:15:23  <Acis.BS_Curve object at 0x7f078ccce500>

Is there a recommendation on how to handle this, or is it dealt with in a way I don't appreciate?

Thank you!
mconsidine

marcocecchiscmgroup, 2023-09-19:
Hello,

including the file typically helps.

Thanks.

jklgrasso, 2024-02-14:
Same here. ^ .f3d is not supported here. Make it a zip.

[parts-tray v5.zip](https://github.com/jmplonka/InventorLoader/files/14281139/parts-tray.v5.zip)

jmplonka, 2024-12-19:
it wasn‘t exact_int_cur. something else…

luzpaz, 2025-05-15:
Can be closed ?