jmplonka/InventorLoader #83: Ipt to Step is not working.

leoheck, 2024-02-26:
Hi, I just found this workbench to convert a model to step.
It looks like it is not working for me. I am not sure if this is a problem with my ipt model or a problem with the InventorLoader.
I am going to share the output of Freecad, here.

I selected to import the step, but it looks like it is importing all the options presented in the popup window.

```
20:39:25  Importing: /home/lheck/Downloads/Caja_Metalica_Octagonal.ipt
20:40:06  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/lheck/.local/share/FreeCAD/Mod/InventorLoader/./importerIL.py", line 118, in insert
    reader.create3dModel(group, doc)
  File "/home/lheck/.local/share/FreeCAD/Mod/InventorLoader/./Import_IPT.py", line 279, in create3dModel
    convertModel(root, doc.Name)
  File "/home/lheck/.local/share/FreeCAD/Mod/InventorLoader/./importerSAT.py", line 151, in convertModel
    ImportGui.insert(stepfile, docName)
<class 'OSError'>: Cannot read STEP file: /home/lheck/Downloads/Caja_Metalica_Octagonal_ipt/0016.step
```

I am also attaching the file here.
[Caja_Metalica_Octagonal.zip](https://github.com/jmplonka/InventorLoader/files/14412468/Caja_Metalica_Octagonal.zip)

I am using the latest Freecad, 0.22

```
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36117 (Git) AppImage
Build type: Release
Branch: main
Hash: 8a3407aee6dff8218282d9cdc3f4e32e7ad958d4
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * InventorLoader 1.4.0
```

leoheck, 2024-04-04:
Ah, ok.

The generated file is this binary. 
[image]

[image]

marcocecchiscmgroup, 2024-04-04:
This is strange, as it should be ASCII.

Attached.
[0016.zip](https://github.com/jmplonka/InventorLoader/files/14872715/0016.zip)

leoheck, 2024-04-04:
I was updating my last message, sorry, hit send before it was ready.

leoheck, 2024-04-04:
The file you shared is good. See:

[image]