jmplonka/InventorLoader #84: FreeCAD 0.21.2 Linux x86_64 segfaults when trying to import Fusion 360 model

Ghostbird, 2024-03-29:
Full information is [https://gist.github.com/Ghostbird/e9d8f1168581b3a7f371c8213294f91e]

## Summary

I tried to import [this file]( https://files.printables.com/media/prints/33072/stls/323617_9542cc9e-7ff7-4fb6-94b4-db2ad25add07/pump-it-up-dance-pad.f3d), FreeCAD segfaults probably because of something that goes wrong in libTKMesh.

It fails regardless of whether I select `nativ` mode or conversion to STEP format.

eMPee584, 2024-09-02:
On a current dev appimage with the posted file, the native import does not crash but it generates strange geometry that is not visible unless selecting parts and fitting the view to selection. Reopening the file (revert didn't work) and selecting "convert to STEP" seems to successfully build the model but segfaults in `StepShape_ShellBasedSurfaceModel::NbSbsmBoundary()` after a few steps. Log is here: [freecad-appimage-0.22.0devR38385.2024-09-02.1112h.log](https://github.com/user-attachments/files/16835314/freecad-appimage-0.22.0devR38385.2024-09-02.1112h.log)

luzpaz, 2024-09-03:
@Ghostbird better to work on development release 0.22.x  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

jmplonka, 2024-12-12:
FreeCad 1.0.0 craches as well on Mac, when tryin to read generated file `BREP.a0b5b80b-1445-44d7-852c-00a145fb5f44.step`
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Fcmcua 1.0.0
  * SearchBar 1.0.2
  * InventorLoader 1.5.1
  * fcgear 1.0.0
  * Importer3D
  * A2plus 0.4.63
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.1
  * Assembly4 0.50.7
  * Assembly3 0.12.2

closed 2026-10-12 by Ghostbird