jmplonka/InventorLoader #86: SAT (and potentially other) import errors in 1.0.0RC1

g3grau, 2024-09-16:
Hi,

the upcoming FC 1.0.0 moved to Python 3.11 where the import (e.g. of a SAT file) fails on open(..., 'rU') because the deprecated opion U has been removed. The fix should be simply to remove the U from all file open commands. Read() should handle both UNIX and DOS line styles.

```
10:04:40  Importing: /home/ggrau/Downloads/M-QFN48W.4.sat
10:04:40  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/home/ggrau/.local/share/FreeCAD/Mod/InventorLoader/./importerIL.py", line 113, in insert
    reader = read(filename)
             ^^^^^^^^^^^^^^
  File "/home/ggrau/.local/share/FreeCAD/Mod/InventorLoader/./importerIL.py", line 46, in read
    if (importerSAT.readText(filename)):
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ggrau/.local/share/FreeCAD/Mod/InventorLoader/./importerSAT.py", line 160, in readText
    with open(fileName, 'rU') as file:
         ^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: invalid mode: 'rU'

``` 

thanks & BR
  Guenter

jmplonka, 2024-12-05:
Could you please share an example (link?)... Would be helpful to speed up bug-fixing.

jmplonka, 2024-12-12:
seems to be a general issue...