jmplonka/InventorLoader #88: UTF decode errors importing .F3D

dezanche, 2024-10-29:
I'm getting the following error on pretty much any .F3D file I try to import (the rest don't import anything):
```
14:24:08  Reading: C:\Users\...\Fusion360\Admin Project\Fabricate ready\Full 3D model_v16.f3d
14:24:08  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 147, in open
    _open(filename, skip, only, root)
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 127, in _open
    reader = read(filename)
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 70, in read
    if (importerF3D.read(filename)):
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerF3D.py", line 144, in read
    folder = read_manifest(f3d, 'Manifest.dat')
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerF3D.py", line 107, in read_manifest
    folder, i = getLen32Text16(data, i) # -> folder name
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerUtils.py", line 895, in getLen32Text16
    txt = data[i: end].decode('UTF-16LE')
  File "C:\Program Files\FreeCAD 0.21\bin\lib\encodings\utf_16_le.py", line 16, in decode
    return codecs.utf_16_le_decode(input, errors, True)
<class 'UnicodeDecodeError'>: 'utf-16-le' codec can't decode byte 0x01 in position 5420: truncated data>
```
Any ideas?

jmplonka, 2024-12-05:
Let me check it out, but therefore I need an example, seems to be a Fusion-Version issue.
Could you share the file?

dezanche, 2024-12-05:
Thanks, here is a file that gives that error 
[Full 3D model_v16.zip](https://github.com/user-attachments/files/18028012/Full.3D.model_v16.zip)
(zipped otherwise it won't let me upload it); I have others from the same project that don't give an error but don't import anything.
This is all under FreeCAD 0.21.1

jmplonka, 2024-12-06:
but there still remains an AccessViolation while converting model.