jmplonka/InventorLoader #90: IPT to Step import error msg

Bwanna, 2024-12-03:
Hi,
Not sure if this is something I'm doing, but opening the following .ipt and converting it to .step produces the following Report.
[Freecad Import Report.txt](https://github.com/user-attachments/files/17997407/Freecad.Import.Report.txt)

Here's the Python Console output:
[Freecad Python Console output.txt](https://github.com/user-attachments/files/17997431/Freecad.Python.Console.output.txt)

Here's the file:
[Horn Button V1.zip](https://github.com/user-attachments/files/17997440/Horn.Button.V1.zip)

Thanks

luzpaz, 2024-12-04:
Please paste your full [About](https://wiki.freecad.org/About) info

Bwanna, 2024-12-04:
> Please paste your full [About](https://wiki.freecad.org/About) info

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * InventorLoader 1.4.0
  * sheetmetal 0.5.8

jmplonka, 2024-12-05:
The import fails due to unsupported Inventor Version 2024. Currently in progress.