jmplonka/InventorLoader #92: ezdxf dependancy undeclared in freecad

ksamak, 2024-12-15:
i saw an error when importing a .f3d in the console:

11:25:22  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ksamak/.local/share/FreeCAD/Mod/InventorLoader/./importerIL.py", line 8, in <module>
    import os, sys, FreeCAD, FreeCADGui, importerSAT, importerDXF, Import_IPT, importerF3D
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ksamak/.local/share/FreeCAD/Mod/InventorLoader/./importerDXF.py", line 8, in <module>
    from ezdxf         import readfile
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'ezdxf'

So i just installed the module. It works now.
i guess it should be declared as dependancy before installing the bench
Thanks for your works!

sys:  Debian 6.1.119-1 stable

luzpaz, 2024-12-15:
What version of FreeCAD and Inventor Loader addon are you using ? 
Paste full [About](https://wiki.freecad.org/About) info please

jmplonka, 2024-12-18:
This could happen, if an earlyer version was used. Please try to uninstall and than reinstall the plugin.