jmplonka/InventorLoader #96: Broken in freecad 1.0

madushan1000, 2025-03-25:
I'm using the flatpak freecad version 1.0 and looks like there has been some api change which breaks `.f3d` import. 
I had to do this change to get it working. 

```diff
diff --git a/importerUtils.py b/importerUtils.py
index ae4785e..775a053 100644
--- a/importerUtils.py
+++ b/importerUtils.py
@@ -306,7 +306,7 @@ def chooseImportStrategyAcis():
 	result = msgBox.exec_()
 	QApplication.restoreOverrideCursor()
 
-	resultMapping = {0:STRATEGY_STEP, 1: STRATEGY_SAT, 2:STRATEGY_CANCEL}
+	resultMapping = {2:STRATEGY_STEP, 3: STRATEGY_SAT, 4:STRATEGY_CANCEL}
 	strategy = resultMapping[result]
 	setStrategy(strategy)
 	return strategy
```

albfan, 2025-04-23:
Maybe there's some access to freecad version on python API to detect freecad version to control this

luzpaz, 2025-05-15:
Maybe through the addon manager ?