jmplonka/InventorLoader #97: Error importing IPT, Freecad 1.0

apper002, 2025-03-26:
01:48:39  Importing: 23801-Black.ipt
01:48:55  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\kurt\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 118, in insert
    reader.create3dModel(group, doc)
  File "C:\Users\kurt\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\Import_IPT.py", line 303, in create3dModel
    convertModel(root, doc.Name)
  File "C:\Users\kurt\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerSAT.py", line 149, in convertModel
    ImportGui.insert(stepfile, docName)
<class 'OSError'>: Access violation

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.8, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * InventorLoader 1.5.1
  * sheetmetal 0.7.22
[/code]

plainpylut, 2025-05-03:
I'm having a similar issue trying import a '.f3d' (Fusion 360 file.  I hope this helps:

FreeCad Version Info:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.3898/1.5
Installed mods: 
  * freecad.gears 1.3.0
  * InventorLoader 1.5.1
  * sheetmetal 0.7.22

FreeCad Python Console I/O (manual line by line execution):
Python 3.11.12 | packaged by conda-forge | (main, Apr 10 2025, 22:09:00) [MSC v.1943 64 bit (AMD64)] on win32
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Start_Start',0)
>>> # Gui.runCommand('Std_Workbench',9)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> ### Begin command Std_New
>>> App.newDocument()
>>> # App.setActiveDocument("Unnamed")
>>> # App.ActiveDocument=App.getDocument("Unnamed")
>>> # Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> # Gui.activeDocument().activeView().viewDefaultOrientation()
>>> ### End command Std_New
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> App.activeDocument().addObject('PartDesign::Body','Body')
<body object>
>>> App.ActiveDocument.getObject('Body').Label = 'Body'
>>> App.ActiveDocument.recompute()
9
>>> import importerIL
>>> from freecad import module_io
>>> module_io.OpenInsertObject("importerIL", "C:/Users/ggkra/3D Objects/Mick/Test.f3d", "insert", "Unnamed")
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\ggkra\FreeCAD_weekly-builds-41264-conda-Windows-x86_64-py311\bin\Lib\site-packages\freecad\module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
  File "C:\Users\ggkra\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 113, in insert
    reader = read(filename)
             ^^^^^^^^^^^^^^
  File "C:\Users\ggkra\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerIL.py", line 70, in read
    if (importerF3D.read(filename)):
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ggkra\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerF3D.py", line 168, in read
    folder = read_manifest(f3d, 'Manifest.dat')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ggkra\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerF3D.py", line 121, in read_manifest
    v, i = getUInt32(data, i)
           ^^^^^^^^^^^^^^^^^^
  File "C:\Users\ggkra\AppData\Roaming\FreeCAD\Mod\InventorLoader\.\importerUtils.py", line 591, in getUInt32
    val, = UINT32(data, offset)
           ^^^^^^^^^^^^^^^^^^^^
struct.error: unpack_from requires a buffer of at least 681836892 bytes for unpacking 4 bytes at offset 681836888 (actual buffer size is 549)
>>> # Gui.runCommand('Std_About',0)
>>> 

[Test.zip](https://github.com/user-attachments/files/20024857/Test.zip)

I hope this helps debug the issue.

Thanks for writing this very useful tool!

Garry Kraemer

luzpaz, 2025-05-15:
soft bump